SequentMicrosystems/megaind-rpi #1: node-red drivers don't install

cmc0619, 2020-11-21:
node-red error on startup:
Nov 21 17:44:33 raspberrypi Node-RED[29573]: 21 Nov 17:44:33 - [warn] [node-red-contrib-sm-ind/bas] Error: Could not locate the bindings file. Tried:
Nov 21 17:44:33 raspberrypi Node-RED[29573]:  → /home/pi/megaind-rpi/node-red-contrib-sm-ind/node_modules/i2c-bus/build/i2c.node
Nov 21 17:44:33 raspberrypi Node-RED[29573]:  → /home/pi/megaind-rpi/node-red-contrib-sm-ind/node_modules/i2c-bus/build/Debug/i2c.node
Nov 21 17:44:33 raspberrypi Node-RED[29573]:  → /home/pi/megaind-rpi/node-red-contrib-sm-ind/node_modules/i2c-bus/build/Release/i2c.node
Nov 21 17:44:33 raspberrypi Node-RED[29573]:  → /home/pi/megaind-rpi/node-red-contrib-sm-ind/node_modules/i2c-bus/out/Debug/i2c.node
Nov 21 17:44:33 raspberrypi Node-RED[29573]:  → /home/pi/megaind-rpi/node-red-contrib-sm-ind/node_modules/i2c-bus/Debug/i2c.node
Nov 21 17:44:33 raspberrypi Node-RED[29573]:  → /home/pi/megaind-rpi/node-red-contrib-sm-ind/node_modules/i2c-bus/out/Release/i2c.node
Nov 21 17:44:33 raspberrypi Node-RED[29573]:  → /home/pi/megaind-rpi/node-red-contrib-sm-ind/node_modules/i2c-bus/Release/i2c.node
Nov 21 17:44:33 raspberrypi Node-RED[29573]:  → /home/pi/megaind-rpi/node-red-contrib-sm-ind/node_modules/i2c-bus/build/default/i2c.node
Nov 21 17:44:33 raspberrypi Node-RED[29573]:  → /home/pi/megaind-rpi/node-red-contrib-sm-ind/node_modules/i2c-bus/compiled/12.19.1/linux/arm/i2c.node
Nov 21 17:44:33 raspberrypi Node-RED[29573]:  → /home/pi/megaind-rpi/node-red-contrib-sm-ind/node_modules/i2c-bus/addon-build/release/install-root/i2c.node
Nov 21 17:44:33 raspberrypi Node-RED[29573]:  → /home/pi/megaind-rpi/node-red-contrib-sm-ind/node_modules/i2c-bus/addon-build/debug/install-root/i2c.node
Nov 21 17:44:33 raspberrypi Node-RED[29573]:  → /home/pi/megaind-rpi/node-red-contrib-sm-ind/node_modules/i2c-bus/addon-build/default/install-root/i2c.node
Nov 21 17:44:33 raspberrypi Node-RED[29573]:  → /home/pi/megaind-rpi/node-red-contrib-sm-ind/node_modules/i2c-bus/lib/binding/node-v72-linux-arm/i2c.node

I also notice that the /megaind-rpi/node-red-contrib-sm-*/node_modules/bindings/package.json files have different pathing between megaind and megabas (which installs fine) ) but I don't know what they're supposed to look like.  But assuming they should be the same and they aren't

cmc0619, 2020-11-22:
I was able to resolve this by copying the i2c-bus/build folder from megabas node-red driver.   Looks like this isn't being done on the megaind side

cp -R ~/megabas-rpi/node-red-contrib-sm-bas/node_modules/i2c-bus/build ~/megaind-rpi/node-red-contrib-sm-ind/node_modules/i2c-bus/build

Obviously this isn't a real fix but workaround for anyone who comes along before this is resolved in the git code

alexburcea2877, 2020-11-22:
Thank you for finding the quick fix. Is strange, the npm should handle all the dependencies.
We will work on solving that.

cmc0619, 2020-11-22:
Also package.json has what I think is a typo.
 "node-red" : {
    "nodes": {
        "bas": "ind.js" <- Should this be ind and not bas?

alexburcea2877, 2020-11-22:
Solve the name bug.
Can you update the repository, delete the "node-modules" folder and reinstall?
Worked on my setup but was worked before too.
Thanks.

cmc0619, 2020-11-22:
That fixed it! Thank you.

closed 2020-11-22 by cmc0619