SequentMicrosystems/megaind-rpi #2: Mega Ind V1.1 CAN protocol

Maudal78, 2020-11-26:
Dear, 
are there any writed  python CAN lib for MegaInd to comunicate with CAN devices?
Were can I find?
Any example?
Thanks
Regards

alexburcea2877, 2020-11-26:
Hi,

The current firmware version does not support CAN, A new firmware version
and python lib to support CAN is under construction.
Regards

Sebastien-GG, 2021-02-09:
Dear Alex
Do you plan to release soon new Firmware with CAN bus update ?
Thanks
Regards

alexburcea2877, 2021-02-09:
Hi,

We don't have the time for finishing the implementation but I think in a month or so we will have the first version.

Regards

Sebastien-GG, 2021-03-16:
Dear Alex 
In the event of you worked on CAN bus update, i m still interested in it !
Regards

Sebastien-GG, 2021-09-02:
Dear ALex
Have you had time to do Can bus firmware update ?
Regrads

alexburcea2877, 2021-09-02:
Not yet!

În joi, 2 sept. 2021 la 12:48, Sebastien-GG ***@***.***> a
scris:

alexburcea2877, 2026-02-14:
No, and because we run into some problems, I don't think we will support CAN on these old cards, sorry.