SequentMicrosystems/megaind-rpi #3: WDT_MAX_POWER_OFF_INTERVAL not defined

gaynoraj19, 2021-03-11:
https://github.com/SequentMicrosystems/megaind-rpi/blob/master/python/megaind/__init__.py
Line 407
```
def wdtSetOffInterval(stack, val):
    ret = 1
    hwAdd = checkStack(stack)
    if 10 > val or val > WDT_MAX_POWER_OFF_INTERVAL:
```

WDT_MAX_POWER_OFF_INTERVAL is not defined

alexburcea2877, 2021-03-11:
Hi,

Thank you for finding the bug.
Update your repository and reinstall the library.
Let me know if is solved.

Alex

gaynoraj19, 2021-03-11:
Will do, thank you. I will post any findings here

alexburcea2877, 2021-03-11:
You are write, thank you. I was planning to write some unit test for checking every function but was unable to do it.

Thanks again.

alexburcea2877, 2021-03-11:
I did the changes, let me know if is something else wrong.

gaynoraj19, 2021-03-11:
It seems to be working correctly now, thank you

closed 2021-03-11 by gaynoraj19