SequentMicrosystems/megaind-rpi #5: Codesys driver?

epgodoy, 2021-04-25:
Hi,
Is there any plans for a codesys driver for this new MegaInd boards (such as the existing one for the old MegaIO-Ind)?
Tks

alexburcea2877, 2021-04-25:
Hi,
There are some plans for Codesys driver but no time frame alocated.

01PetruS, 2023-01-27:
Hello, 
Is there any way to use those cards in Codesys?
Thank you,
Petru

alexburcea2877, 2023-01-27:
Hi,
Unfortunately the situation was not changed since last time.
If you wish to do the driver and share it, we give you all the support you need (I2C address map, parameters scale factors and measurement units, etc) and one free Industrial card.
Or you can wait for us but as you can see, we are too busy redesign because component shortages and releasing new cards at customers requests, to do it in a reasonable time interval. 
Thank you for your understanding,
Alex.

01PetruS, 2023-01-27:
Hello,  
 Yes, i can understand-this components shortages is a real problem and it is affecting all the industries.
I can start to take a look on it and do some tests.
Just let me know what do i need to do to go further. 
Much appreciate your help on this. 
Petru

alexburcea2877, 2023-01-27:
Hi, 
Send me an email to olcitu@gmail.com so I can respond you with the design files for the BAS codesys driver so you have it as reference. Unfortunately the devicedescriptor.xml and license files are corrupted so you have to recreate them. 
I wait for your email.
Alex.

alexburcea2877, 2024-03-24:
We have done the Codesys [driver](https://github.com/SequentMicrosystems/SM_CODESYS/tree/main/Industrial_Automation), if you are still interested.

closed 2024-03-24 by alexburcea2877