SequentMicrosystems/megaind-rpi #6: sudo make install fails

transitvanman, 2021-06-16:
I have a RPI4 with clean Ubuntu 21.04.
After installing 4relind and smartfan, megaind-rpi failed sudo make install. However the python libs and node-red work fine. Unfortunately I need to calibrate the outputs and would like to use the cli please. 

I had no trouble a month ago on Raspberry OS. sudo make install worked on all three repo's i needed.

Here are the last lines of the terminal output:

/usr/bin/ld: src/dout.o:(.rodata+0x6c8): multiple definition of `CMD_UOUT_CAL_RST'; src/megaind.o:(.rodata+0x6c8): first defined here
/usr/bin/ld: src/dout.o:(.rodata+0x700): multiple definition of `CMD_IOUT_CAL_RST'; src/megaind.o:(.rodata+0x700): first defined here
/usr/bin/ld: src/dout.o:(.data.rel.ro.local+0x0): multiple definition of `CMD_DOD_READ'; src/megaind.o:(.rodata+0x738): first defined here
/usr/bin/ld: src/dout.o:(.data.rel.ro.local+0x38): multiple definition of `CMD_DOD_WRITE'; src/megaind.o:(.rodata+0x770): first defined here
/usr/bin/ld: src/dout.o:(.data.rel.ro.local+0x70): multiple definition of `CMD_LED_READ'; src/megaind.o:(.rodata+0x7a8): first defined here
/usr/bin/ld: src/dout.o:(.data.rel.ro.local+0xa8): multiple definition of `CMD_LED_WRITE'; src/megaind.o:(.rodata+0x7e0): first defined here
collect2: error: ld returned 1 exit status
make: *** [Makefile:22: megaind] Error 1

alexburcea2877, 2021-06-16:
Hi,
Can you update the repository and try again?

~$ cd megaind-rpi/  
~/megaind-rpi$ git pull
~/megaind-rpi$ sudo make clean
~/megaind-rpi$ sudo make install

alexburcea2877, 2021-06-16:
Now is fixed, please update, reinstall and let me know if is working.

transitvanman, 2021-06-17:
Thanks mate. Followed your instruction and its all good.

Good work.

closed 2021-06-17 by transitvanman