SequentMicrosystems/megaind-rpi #8: Need Application help for Ind. Automation HAT for RPI3 as IOTEdge

Controlnetix, 2021-09-09:
Hello,

Here is our system setup.
We are planning to use Ind. Automation HAT with Raspberry pi3 (OS: Raspbian Stretch). 

PI3 is acting as an azure iot edge device.  Azure iot edge runtime on pi3 manages-executes existing custom edge modules built in Visual Studio C#.

To use Ind. HAT in our application, we can probably build a new edge module ether in C#/Python OR modify existing module (C#) to reach out to Ind HAT.   
 
Would you please suggest a possible ways/options to effectively Integrate Ind. HAT in above setup?

alexburcea2877, 2021-09-09:
Hi,
The fastest way is to use the python [library ](https://github.com/SequentMicrosystems/megaind-rpi/tree/master/python) we provide.
Unfortunately we have limited experience with C# but if you wish to make your own "megaind" library for C# we can help with the low level stuff.

Controlnetix, 2021-09-10:
Thanks Alex for your reply and suggestions to use python library.

Would you please help with following? Setup script exited with error (error 2).
We are trying to build python iotedge module so we can use python library. Dev box using VS Code (on W10 Pro 64bit).
We have following error while completing setup step from https://github.com/SequentMicrosystems/megaind-rpi/tree/master/python
Here's out put from vs code terminal window: 
##############
PS C:\Users\Admin\Documents\VSCode\EdgeSolution> **git clone https://github.com/SequentMicrosystems/megaind-rpi.git**
Cloning into 'megaind-rpi'...
Receiving objects: 100% (249/249), 463.61 KiB | 5.94 MiB/s, done.
Resolving deltas: 100% (129/129), done.5)
remote: Total 249 (delta 129), reused 50 (delta 23), pack-reused 0
PS C:\Users\Admin\Documents\VSCode\EdgeSolution> **cd megaind-rpi/python/**
PS C:\Users\Admin\Documents\VSCode\EdgeSolution\megaind-rpi\python> python3 setup.py install
Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.
PS C:\Users\Admin\Documents\VSCode\EdgeSolution\megaind-rpi\python> **python setup.py install** 
running install
running bdist_egg
running egg_info
creating megaind.egg-info
writing requirements to megaind.egg-info\requires.txt
writing megaind.egg-info\PKG-INFO
writing top-level names to megaind.egg-info\top_level.txt
writing dependency_links to megaind.egg-info\dependency_links.txt
writing manifest file 'megaind.egg-info\SOURCES.txt'
reading manifest file 'megaind.egg-info\SOURCES.txt'
writing manifest file 'megaind.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
creating build
creating build\lib
creating build\lib\megaind
copying megaind\__init__.py -> build\lib\megaind
creating build\bdist.win-amd64
creating build\bdist.win-amd64\egg
creating build\bdist.win-amd64\egg\megaind
copying build\lib\megaind\__init__.py -> build\bdist.win-amd64\egg\megaind
byte-compiling build\bdist.win-amd64\egg\megaind\__init__.py to __init__.pyc
creating build\bdist.win-amd64\egg\EGG-INFO
copying megaind.egg-info\PKG-INFO -> build\bdist.win-amd64\egg\EGG-INFO
copying megaind.egg-info\SOURCES.txt -> build\bdist.win-amd64\egg\EGG-INFO
copying megaind.egg-info\dependency_links.txt -> build\bdist.win-amd64\egg\EGG-INFO
copying megaind.egg-info\requires.txt -> build\bdist.win-amd64\egg\EGG-INFO
copying megaind.egg-info\top_level.txt -> build\bdist.win-amd64\egg\EGG-INFO
zip_safe flag not set; analyzing archive contents...
creating dist
creating 'dist\megaind-1.0.3-py2.7.egg' and adding 'build\bdist.win-amd64\egg' to it
removing 'build\bdist.win-amd64\egg' (and everything under it)
Processing megaind-1.0.3-py2.7.egg
Copying megaind-1.0.3-py2.7.egg to c:\python27\lib\site-packages
Adding megaind 1.0.3 to easy-install.pth file

Installed c:\python27\lib\site-packages\megaind-1.0.3-py2.7.egg
Processing dependencies for megaind==1.0.3
Searching for smbus
Reading https://pypi.org/simple/smbus/
Downloading https://files.pythonhosted.org/packages/4d/5c/70e14aa4f0c586efc017e1d1aa6e2f7921eefc7602fc2d03368ff912aa91/smbus-1.1.post2.tar.gz#sha256=f96d345e0aa10053a8a4917634f1dc37ba1f656fa5cace7629b71777e90855c6
Best match: smbus 1.1.post2
Processing smbus-1.1.post2.tar.gz
Writing c:\users\admin\appdata\local\temp\easy_install-r3rhnz\smbus-1.1.post2\setup.cfg
Running smbus-1.1.post2\setup.py -q bdist_egg --dist-dir c:\users\admin\appdata\local\temp\easy_install-r3rhnz\smbus-1.1.post2\egg-dist-tmp-w3aozd
building 'i2c' library
error: Setup script exited with error: [Error 2] The system cannot find the file specified
PS C:\Users\Admin\Documents\VSCode\EdgeSolution\megaind-rpi\python>
##############

alexburcea2877, 2021-09-11:
Looks like is a problem with one of the dependencies. Try to install smbus library separately. I do not know if is working in w10 but run "pip install smbus".

Controlnetix, 2021-09-11:
Pls see the terminal output when run "pip install smbus".
[image]

Controlnetix, 2022-01-04:
> Hi, The fastest way is to use the python [library ](https://github.com/SequentMicrosystems/megaind-rpi/tree/master/python) we provide. Unfortunately we have limited experience with C# but if you wish to make your own "megaind" library for C# we can help with the low level stuff.

Controlnetix, 2022-01-04:
> Hi, The fastest way is to use the python [library ](https://github.com/SequentMicrosystems/megaind-rpi/tree/master/python) we provide. Unfortunately we have limited experience with C# but if you wish to make your own "megaind" library for C# we can help with the low level stuff.

How do we get started with megaind c# library? What do you need from us?

alexburcea2877, 2022-01-04:
Hi,
Can you provide the exactly the os, and the IDE for writing code you use (eventually a documentation about installing it on raspberry) so I can make a similar system so I can really help you with the drivers for the megaind?

Controlnetix, 2022-01-04:
> 
Alex Please see details about os and ide. 
Dev Box:
Windows 10 Pro, 10.0.19044 Build 19044, x64
IDE: Visual Studio Pro 2019, Ver 16.11.8 
Project Template: Azure IoTEdge (Linux arm32v7), C#

Thanks
Nilesh Suthar