SequentMicrosystems/megaind-rpi #11: 'setLed' called by python/tests/flash_leds.py not defined

jderoo, 2021-12-14:
Cloned the project today and ran through the setup.  Was able to run the other python tests, but not the flash_leds.py test.
```
pi@fourtune:~/megaind-rpi/python/tests $ sudo python3 flash_leds.py -v
Traceback (most recent call last):
  File "/home/pi/megaind-rpi/python/tests/flash_leds.py", line 24, in <module>
    flash_seq(10)
  File "/home/pi/megaind-rpi/python/tests/flash_leds.py", line 8, in flash_seq
    ind.setLed(stack, j + 1, 1)
AttributeError: module 'megaind' has no attribute 'setLed'
pi@fourtune:~/megaind-rpi/python/tests $ 
```

alexburcea2877, 2021-12-15:
Hi,

Sorry, we forgot to commit the last version of the python library.
Please reinstall the library and try again.

Best regards,

În mie., 15 dec. 2021 la 01:26, John DeRoo ***@***.***> a
scris:

> Cloned the project today and ran through the setup. Was able to run the
> other python tests, but not the flash_leds.py test.
>
> Traceback (most recent call last):
>   File "/home/pi/megaind-rpi/python/tests/flash_leds.py", line 24, in <module>
>     flash_seq(10)
>   File "/home/pi/megaind-rpi/python/tests/flash_leds.py", line 8, in flash_seq
>     ind.setLed(stack, j + 1, 1)
> AttributeError: module 'megaind' has no attribute 'setLed'
>
> <https://github.com/SequentMicrosystems/megaind-rpi/issues/11>, or
> unsubscribe
> or Android

jderoo, 2021-12-15:
Alex,

Thanks - that worked - all better now.

John DeRoo

closed 2021-12-15 by jderoo