SequentMicrosystems/megaind-rpi #16: I2C Command frequency

vtx22, 2022-05-04:
Hello,

I've written a C++ application that controls the Industrial Automation Board via I2C. Sending the commands manually works fine so far, however I've noticed some glitches when sending fast consecutive commands. Introducing a delay between the commands fixed the problems somewhat. The delay came out to be about 8 - 10ms through experimentation. Is that observation about right or is the fastest possible I2C command frequency higher? Because 10ms delay sounds quiet high in my opinion.

Regards

alexburcea2877, 2022-05-05:
Hi,
Let me know the firmware version of your card (run the "megaind 0 board" command in the terminal)

vtx22, 2022-05-05:
Hi,
the firmware version is 01.10

alexburcea2877, 2022-05-05:
Please update to the latest, instructions: https://github.com/SequentMicrosystems/megaind-rpi/blob/master/update/README.md
Let me know if something change.

vtx22, 2022-05-05:
Thanks, that decreased the minimum time to about 1ms, which is fast enough for my application. However, i've noticed some odd glitches. As a test, i wrote a program that toggles the four on board LEDs every second by sending the four I2C commands consecutively with 1ms delay (as you can see in the picture)

[image]

In about 95% of the time everything goes right, but sometimes one LEDs does not toggle. The same behaviour occurs when changing the delay to 2ms, as if the IABoard is sometimes busy and cannot react on the command. Is there a way to work around that issue (instead of increasing the delay again) or should i just double ckeck with a read command if the right value was set?

alexburcea2877, 2022-05-05:
The i2c bus respond faster than millisecond but the loop that check if a new LED command has arrived repeat every millisecond (inside the microcontroller on the card).
Try to use the 2ms delay, that should be enough for the LED command to be executed.

vtx22, 2022-05-05:
I see, thanks for your help

closed 2022-05-05 by vtx22