SequentMicrosystems/megaind-rpi #20: Python libary - function "get4_20In(0, 1)" wrong read if Input Signal is negativ (nothing connected)

Csepp, 2022-08-16:
If the Current Input Signal 4-20mA is below 0 eg. -0,001 mA the Pyton funtion returns back 65,4mA. Something on the conversion in the Python code is wrong. 

When using the Bash code the reading is correct. 

Could somebody please check this?
Peter

alexburcea2877, 2022-08-16:
Hi,
Thanks for pointing this out, I will fix it and let you know.

Alex.

alexburcea2877, 2022-08-17:
Hi
Can you tell me your card hardware version? Is printed on the card.

Csepp, 2022-08-17:
Dear Alex,
the Hardware is MEGA-IND2-V2,
Pete

alexburcea2877, 2022-08-17:
Hi,

A new firmware version was created to address the problem, please update your card firmware version (last firmware version for your hardware version is 1.14).
Please follow the instructions from here: https://github.com/SequentMicrosystems/megaind-rpi/tree/master/update
Let me know the results.
Alex.

Csepp, 2022-08-17:
Dear Alex, 
I started the update. It did tell me the FW is up to date - but it was not. I checked, before update i had v 1.13 - afterwards, 1.1400000000000001. I can test it tomorrow - need to do a new calibration. Peter

pi@raspberrypi:~/megaind-rpi/update$ ./update 0
You are ready to update the firmware on your board.
Your Raspberry Pi must be connected to the internet.
We strongly recommend to disconnect all outputs from the board since they can change state unpredictably.
Type "yes" (without quotes) to continue or anything else to quit:
yes
CPUID: 480034000857415638363820
Downloading firmware...               Board hardware version < 3.0
Done
Writing flash...100%         Done
pi@raspberrypi:~/megaind-rpi/update$

Version Check:

>>> megaind.getFwVer (0)
1.13
>>> megaind.getFwVer (0)
1.1400000000000001

alexburcea2877, 2022-08-17:
The calibrations are not erased by the firmware update.

Csepp, 2022-08-18:
Dear Alex, 
Thank you it works now. No values anymore blow 0. Great. 
Have a great day,
Peter

alexburcea2877, 2022-08-18:
Hi 
Thank you for reporting the bug report.

Alex.

closed 2022-08-18 by alexburcea2877