SequentMicrosystems/megaind-rpi #23: How to write/read from RS485

biohazduck, 2023-04-06:
Hello! I've been perusing your library and was wondering how would one write bytes over the RS485 from the raspberry Pi? I can see there is a setup call in https://github.com/SequentMicrosystems/megaind-rpi/blob/master/src/rs485.c. There is also a way to recover the values of the settings. I cannot tell how to write a series of bytes.

alexburcea2877, 2023-04-06:
The driver for the Industrial automation just configure the card local processor serial port. There are 2 mode for using the RS485 interface on the card: 1) The local processor use  it to implement Modbus RTU Slave so you can access the card remotely without the raspberry.
2) The local processor free the lines so you can place the TX and RX jumpers and use the interface with the raspberry UART port.
For the second mode of operation you access serial0 of the raspberry and there are tons of examples drivers, libraries using all kind of  languages so no need for us to repeat them.

alexburcea2877, 2023-07-03:
The configuration is good for using the Raspberry UART port with the RS485 transceiver.
Are you sure you set the right baud rate for the port?
Can you try to place a 120 ohms resistor (terminator) in parallelly to the RS485 bus?

biohazduck, 2023-07-03:
a 120ohm resistor between A and B?

alexburcea2877, 2023-07-03:
Yes

biohazduck, 2023-07-03:
With 138ohm parallel resistor I don't manage to read anything from bus.
```
$ python3 modbusrtu_test.py 
cmd result: Invalid MODBUS device address: [1, 255]! Set to 1.
done

try address 10
trying register 7939
No communication with the instrument (no answer)
```

This same code works when switching to a RS485 to USB transceiver.

biohazduck, 2023-07-03:
Worth mentioning I get the same result when connecting the jumper next to the RS485 connector.

alexburcea2877, 2023-07-03:
First, you do not need to disable the Modbus interface of the card every time you try to use the port. The settings are saved in the flash so only once you have to do this.
Maybe this is the problem.. can you read back the modbus settings (megaind 0 rs485rd) and share the output?

biohazduck, 2023-07-03:
```
$ megaind 0 rs485rd
<mode> <baudrate> <stopbits> <parity> <add> 0 9600 1 0 1
pi@testbedx:~ $ megaind 0 rs485wr 0 0 0 0 0
Invalid MODBUS device address: [1, 255]! Set to 1.
done
pi@testbedx:~ $ megaind 0 rs485rd
<mode> <baudrate> <stopbits> <parity> <add> 0 9600 1 0 1
```

biohazduck, 2023-07-03:
Signal output on B RS485 with respect to ground: With jumper (first) near the RS485 port and without (second):
[image]
[image]

alexburcea2877, 2023-07-03:
Read from the device is good, actually just the first parameter counts to be zero (card modbus interface disabled) the rest of the parameters are ignored for this case. Is useless to keep setting this because is saved in flash.
The RS485 is a differential signal so you have to measure between A and B or measure both respect to the GND and substract them (one from another, some oscilloscopes have this function).
Can you put common GND between ind card and the device you are talking with?
Your situation is strange because we test each card after programming and one of the test is to communicate with the card over modbus, it is not exactly your situation but guarantee that the RS485 transceiver is good.

biohazduck, 2023-07-11:
So the issue was the stop bits. The Raspberry Pi is incapable of 2 stop bits or other-than-none parity on the ttyS0 UART due to the hardware (mini-uart).

alexburcea2877, 2023-07-11:
You are welcome.

closed 2023-07-11 by alexburcea2877