SequentMicrosystems/megaind-rpi #24: Changing the I2C Address

ajaymccollum, 2023-05-08:
I'm using a Rockpi 4C+ from OKdo. There doesn't seem to be an i2c-1 like on a regular rpi. I have i2c-0,6,7 and 9. Is there a way to change which address (sorry, not sure if that's the right term) short of going in and modifying your .js file?

alexburcea2877, 2023-05-08:
Hi,
That is the I2C port you talking to.  We plan to make this settable from node menu but until then you can change the port manually. Go to the "/.node-red/node_modules/node-red-contrib-sm-ind/" folder and edit the "ind.js" file. Replace the "I2C.openSync(1);" sentence with "I2C.openSync(7);" everywhere in the file (you should find him 14 times). Enable the i2c-7 port: Point 3 of [Sequent Microsystems faq](https://sequentmicrosystems.com/apps/faqs?faq-section-id=71090&faq-article-id=325901) Question "Can I use Sequent Microsystems I/O cards with ROCK PI 4 ?"

ajaymccollum, 2023-05-08:
Thank you for the fast response. I had tried that before, but ran into the error "Error: EACCES: permission denied, open '/dev/i2c-7'". I hadn't looked very much into it because I wasn't sure if changing that value was even a fix.
I know this isn't your issue, but I'm wondering if you happen to know the fix for that. I'll go ahead and close this issue either way.

alexburcea2877, 2023-05-08:
Did you enable the i2c-7 port? (sudo nano boot/hw_intfc.conf   | turn on i2c7)

ajaymccollum, 2023-05-09:
I do; I enabled it through "rsetup". When I go to edit that file, there's a note in there about how it's deprecated, and to use "rsetup". But if I install libmraa, then it seems to use the hw_intfc.conf file. 
I've enabled my I2C ports through both methods and after changing the port used in your .js file, it still comes up "permission denied".
Right now, the damned thing seems to have crashed (this is happening a lot, can't imagine the reason) and I can't get into any sort of recovery, so I've been starting from scratch over and over....
It seems I got bigger problems with this now, so I'll leave you alone, and thank you very much for the quick responses and help!

closed 2023-05-09 by ajaymccollum