SequentMicrosystems/megaind-rpi #26: installing node-red from repo: MODULE_NOT_FOUND

PattonDynamics, 2023-09-10:
When I install node-red-contrib-sm-ind from a local repo, it installs, but does not show up in the node-red palette. Under user setting, it is listed along with all of the nodes, but it has a "MODULE_NOT_FOUND" warning. None of the nodes are enabled. 

If I remove it, and re-install it directly from node-red, then it works properly.

The only two differences I have identified are
1. the node_modules folder node-red-contrib-sm-ind is a symbolic link when installed locally. 
2.  the dependency listing in the package.json.
    - local install: "node-red-contrib-sm-ind": "file:../repos/megaind-rpi/node-red-contrib-sm-ind",
    - Install from node-red: "node-red-contrib-sm-ind": "~1.0.4",

The version shown is 1.04. in both cases.
I am installing on a RaspberryPi

Not the end of the world, but frustrating.

alexburcea2877, 2023-09-10:
Thank you for the bug report.
We will look into it and if we can not solve it we keep only the second installing method.
One small question, have you change your current folder to the node-Red (.node-red) install folder before installing the node, as in the instructions?
Can also provide node-red and npm versions on your Raspberry?

PattonDynamics, 2023-09-11:
Yes, I run the following command from within ~/.node-red
`npm install ~/repos/megaind-rpi/node-red-contrib-sm-ind`

It appears to install correctly, I see everything in the node modules folder. Then I stop/start node-red. None of the sequent nodes are available in the palette, and I see the "MODULE_NOT_FOUND" in the palette settings. 

Very strange, I can't tell why this method does not work, but installing from within node-red works fine.

node-red version: 3.1.0
npm -v: 9.6.7

CodyOakes, 2023-10-17:
Had the same issue. Thanks for the work around @PattonDynamics

calcut, 2025-05-07:
Same issue - at first it wasn't clear to me how to "re-install it directly from node-red" but I used the command `npm pack` to make a tgz file of the module and then was able to install it using the upload tgz button in the palette manager