SequentMicrosystems/megaind-rpi #27: Request for Assistance: Installing "megaind" with PySide6 in a Virtual Environment

AndriesTuinhof, 2023-11-22:
I'm currently working on integrating the "megaind" library with PySide6, and PySide6 necessitates the use of a virtual environment. Nevertheless, I've run into challenges while attempting to install the "megaind" library within the virtual environment. Despite trying several approaches, I haven't been able to make it work successfully. Do you have any suggestions on how to achieve this? Additionally, I'm curious about why the "megaind" project isn't available on PyPI. Having it on PyPI would simplify the installation process by allowing us to use 'pip install'.

Many thanks and kind regards,
Andries

alexburcea2877, 2023-11-22:
We will try to publish it on PyPi and let you know.
I have no experience with virtual environments but normally if you go to the megaind-rpi/python/ folder and run "pip install ." is working, or you can use "python -m pip install ."
Alex.

AndriesTuinhof, 2023-11-22:
Just for your reference, I installed Qt using the instructions from this link: [Qt for Python Quick Start](https://doc.qt.io/qtforpython-6/quickstart.html#quick-start).

Currently, I have a folder where I've installed 'megaind-rpi' and followed the instructions provided in the readme file, which worked perfectly fine except when I activated the virtual environment (venv).

To resolve this issue, I navigated to the './megaind-rpi/python' directory and attempted to install it using either 'pip install .' or 'python -m pip install .'. However, I encountered the 'legacy-install-failure' error. To work around this, I used the '--break-system-packages' flag.

Interestingly, when I remain in the 'megaind-rpi/python' directory, I can use the library without any issues (when i go to the program it doesn't_. However, 'pip list' indicates that it is not installed within the virtual environment.

alexburcea2877, 2023-11-23:
Hi, 

We just published the library on the PyPI you can use "pip install megaind" to install it.

AndriesTuinhof, 2023-11-24:
"I just gave it a try, and it worked flawlessly. Afterward, I simply had to run 'pip install smbus' since that library wasn't initially installed in the environment. Nevertheless, I greatly appreciate your assistance – you're a hero!

closed 2023-11-24 by AndriesTuinhof

alexburcea2877, 2023-11-24:
I am glad you made it, please give us a review on [Sequent Microsystems ](https://sequentmicrosystems.com) if you appreciate our products.
Thank you.