SequentMicrosystems/megaind-rpi #28: Why can't I activate all 4 LEDs at once?

IamSierraCharlie, 2023-12-23:
As the subject suggests.

If I do:
```
import megaind as meg
meg.setLed(0,1,1)
meg.setLed(0,2,1)
meg.setLed(0,3,1)
meg.setLed(0,4,1)
```
Only the first 3 LEDs come on.  I then disable any LED (except the 4th) and the final LED comes on.

Also, if I do 

`meg.setLedAll(0, 1)`

only the first LED comes on - not all.

I updated the firmware and power cycled the pi - no change.

IamSierraCharlie, 2023-12-23:
FW version is 2.06 by the way

alexburcea2877, 2023-12-23:
To update all led's you need to send a bitmap with all 4: "meg.setLedAll(0,15)" will enable all the LED's ;  "meg.setLedAll(0,5)" will turn on LED 1 and 3, and turn off led 2 and 4.

IamSierraCharlie, 2023-12-23:
Thanks,  I'll give it a try!

alexburcea2877, 2024-03-24:
You never answered but I am sure it works

closed 2024-03-24 by alexburcea2877