SequentMicrosystems/megaind-rpi #30: Watchdog reset node

PBnFlash, 2024-03-23:
I understand there are no timing guarantees on a rpi but surely it would make sense to manually trigger the watchdog reset from node-red directly rather than calling an external script?

alexburcea2877, 2024-03-23:
You can do that using exec node, running  wdt command

PBnFlash, 2024-03-23:
I had to shut down the node red service before it would reboot, even after submitting the wdtr command from the command line. Node red doesn't send the command automatically while in development mode or something?

Is there any way to check how long it's been since the last watchdog reset?

alexburcea2877, 2024-03-23:
I am not understand the problem but NodeRed does nothing unless you tell it to.
There is no way to find the time from last watchdog reset.

PBnFlash, 2024-03-24:
Sorry, I was trying to trigger the watchdog and have the device reset automatically (to test the feature.) However even after starting the watchdog from the console the device would not reset with the node red service running.

Feel free to close this I'll examine my setup closer and if I can replicate I'll re-open with replication steps. 

Impressed with the hat so far lovely device.

alexburcea2877, 2024-03-24:
You should close this when you consider is done.
Any access to the card prevents watchdog from reset the PI, it consider the RPI is not hung.

PBnFlash, 2024-03-24:
Ah, interesting. Thank you for your help.

closed 2024-03-24 by PBnFlash