SequentMicrosystems/megaind-rpi #33: Using getOD setOD

Golden923, 2024-07-30:
Hi,

I am a bit new to this but I am trying to using the python commands to get and set output values for a sequent microsystems industrial automation hat. 

I see that it is a sub folder in the depository and so perhaps I need to do something else to use the functions once i import megaind?

alexburcea2877, 2024-07-30:
Hi,

If you imported the megaind library you need to access his functions. Check out the function prototypes in [readme](https://github.com/SequentMicrosystems/megaind-rpi/blob/master/python/README.md).

Golden923, 2024-07-30:
Hi,

yes I did import the megatons library as per the readme, but I see there is a getOd and set of function in a seperate folder in the depository. How can I use these functions?

alexburcea2877, 2024-07-31:
As in the readme example.

import megaind
megaind.setOdPWM(0, 1, 100) 

You will see in the readme what means each parameter.

Golden923, 2024-07-31:
Ahh ok I understand now, we have to toggle it ON/OFF in terms of 0 or 100 in the PWM function. I assumed it must have just been on 0 or 1 input boolean logic.

Thanks again and I appreciate your replies!

closed 2024-07-31 by Golden923