SequentMicrosystems/megaind-rpi #35: 12v fan on 0-10v output

PBnFlash, 2025-05-17:
Hello,

I'm going to power a little (12v 60mm 12mA) case fan on a 0-10v output.
Specs call out 10 mA and 1k ohm loads.

The LM258DT looks like it can take it.

Is the current limit to stay in a nice linear range or is there a reason to not use it for a fan I'm not seeing.

Thanks,

alexburcea2877, 2025-05-17:
Hi,

Because the fan is an inductive load, it is hard to predict what is going to happen, but I think will not blow the opamp.
Even was not designed for this kind of load, you can try it.

Alex.

PBnFlash, 2025-05-17:
Thanks,

And I am slightly concerned with flyback from the fan, but it should be minimal on such a small fan, and I am not using the 10v out for this project so loosing one or two channels isn't too devastating.

I'll go ahead and close this, if I blow the output I'll come back and warn of my hubris.

Regards