SequentMicrosystems/megaind-rpi #36: Open Drain Output - can't change frequency

calcut, 2025-06-16:
I'm trying to change the output frequency of OD2 using megaind command line.
I have an oscilloscope hooked up and can see a frequency of ~6.37kHz, looks to be the maximum.

I try to change the frequency to 1kHz like this:

```
testrig@testrig:~ $ megaind 0 odfwr 2 1000
done
testrig@testrig:~ $ megaind 0 odfrd 2
1000
```
But no change, stuck at max frequency.

To confirm I'm looking at the correct output I can run commands like this and see the duty cycle changing no problem

```
testrig@testrig:~ $ megaind 0 odwr 2 59
done
testrig@testrig:~ $ megaind 0 odwr 2 50
done
```

I also noticed that the tool won't let me change OD4. only 1..3.
Not an issue in my application but it makes me wonder if this feature has escaped testing?
```
testrig@testrig:~ $ megaind -h odfwr
	odfwr:		Write open-drain output PWM frequency value (10..6400Hz)
	Usage:		megaind <id> odfwr <channel[1..3]> <value[10..6400]>
	Example:		megaind 0 odfwr 2 250; Set PWM frequency of 250Hz for open-drain output channel #2 on Board #0

testrig@testrig:~ $ megaind 0 odfwr 4 1000
Invalid Open-drain channel for frequency change, must be 1..3!
```

I have megaind v1.1.8 (latest from github) And I've just updated the firmware on my Industrial Automation HAT (v3)

alexburcea2877, 2025-06-16:
Hi,

Yes, this is a new feature so you have to update the firmware on the card. Because the timer used for channel 4 is also used for features that will not work properly if you change the frequency, we limited the feature for channels 1 to 3

calcut, 2025-06-16:
Hi, thanks for quick reply :-)
I did already update the firmware.
Also I noticed this line in the code 

https://github.com/SequentMicrosystems/megaind-rpi/blob/652448333b5c9d04be1ffd88889b6e10751076aa/src/analog.c#L486

Might indicate that it is hard coded to only write to OD1 not 2 or 3?
However I couldn't get OD1 working either

calcut, 2025-06-16:
ignore that last bit... I see there is a baseadd line that increments it.

alexburcea2877, 2025-06-16:
I ment to make the card firmware update, following the instructions from here: https://github.com/SequentMicrosystems/megaind-rpi/blob/master/update/README.md

calcut, 2025-06-16:
Yep, I used the "ubuntu" firmware update in case that makes a difference.

```
testrig@testrig:~/megaind-rpi/update/ubuntu $ ./update 0
You are ready to update the firmware on your board.
Your Raspberry Pi must be connected to the internet.
We strongly recommend to disconnect all outputs from the board since they can change state unpredictably.
Type "yes" (without quotes) to continue or anything else to quit:
yes
CPUID: 24005e001550344838333820
Downloading firmware...               Board hardware version >= 3.0
Done
Writing flash...100%         Done
```

alexburcea2877, 2025-06-16:
This feature is implemented on cards v4.x, but I guess it will also be doable for 3.2. Let me check on this and get back to you.

alexburcea2877, 2025-06-16:
Hi,

You can update your card firmware and try again. Please let me know the result

alexburcea2877, 2025-06-17:
Ok, I rolled back to 2.10, run the update again to have a functional card.
Give me some time to free my schedule so I can do it properly.

alexburcea2877, 2025-09-26:
Hi,
Unfortunately, there is no news on this. I'm sorry. We've been busy here lately.
The version we currently sell has the frequency change mechanism. It is still on my list to update the V3, but I cannot give you an estimate.